jeelabs/el-client #3: mqtt subscriptions return empty message data

tve, 2016-02-20:
From gitter:

> @BrewerGlyph Ok thanks. I have added a few subscriptions line into the sketch. I got different result for different topic.
```
Received: topic=testing
data={"rssi":-29, "heap_free":19400}
ELC: got 46 @27B: 3 607 2 19 0 2F 72 75 6D 61 68 2F 70 6F 6D 70 61 2F 70 72 65 73 73 75 72 65 48 65 61 64 0 0 0 4 0 33 2E 33 30 49 87
RESP_CB: 607 2
Received: topic=/rumah/pompa/pressureHead
data=
publishing
ELC: got 30 @27B: 3 607 2 B 0 2F 65 73 70 2D 6C 69 6E 6B 2F 31 0 2 0 31 33 0 0 F3 BD
RESP_CB: 607 2
Received: topic=/esp-link/1
data=13
```
>One was blank (topic=/rumah/pompa/pressureHead), the other one was okay (topic=testing)
I use mqtt-spy as a client that also subscribe to the same topics and mqtt-spy received all the data just fine.

And

>A topic of "testing" seems to work fine, but a topic of "testing1" gets to be garbage-y. "testing22" ends up with blank data.

tve, 2016-02-20:
I can repro using:
```
  mqtt.subscribe("/rumah/pompa/#");
```
and
```
    itoa(count+99, buf, 11);
    mqtt.publish("/rumah/pompa/pressureHead", buf);
```
resulting in
```
ELC: got 46 @2A5: 3 649 2 19 0 2F 72 75 6D 61 68 2F 70 6F 6D 70 61 2F 70 72 65 73 73 75 72 65 48 65 61 64 0 0 0 2 0 39 33 0 0 46 E0
RESP_CB: 649 2
Received: topic=/rumah/pompa/pressureHead
data=
```

BrewerGlyph, 2016-03-04:
This appears completely fixed in v2.2beta2... nice work!

closed 2016-12-03 by tve