jeelabs/el-client #4: REST HTTP request to CouchDB problems

tve, 2016-02-20:
Copied from https://github.com/jeelabs/esp-link/issues/106:

@ksuie:
>When I use esp-link(v2.1.7 and v2.2.beta1) REST request to CouchDB(port 5984)
`rest.begin("192.168.2.25", 5984, false); rest.get("/");`
Debug Log:
`REST: status=200, body=0`
>Can't get any response body.

>But REST request to Lighttpd(port 80) is ok.
`REST: status=200, body=15`
Any suggestions?

@tve:
> What do you expect to receive from couchdb in the response? Could it be that the response it too big to fit?
 
@ksuie:
```
$ curl http://192.168.2.25:5984 {"couchdb":"Welcome","uuid":"824b5da6197f7d370930e7a7923abec4","version":"1.6.1","vendor":{"version":"1.6.1","name":"The Apache Software Foundation"}} 
Response size: 150 bytes.
```
>Add receive buffer to 512 bytes
`uint16_t code = rest.waitResponse(response, 512);`
still same problem, REST: status=200, body=0

>I use lighttpd to measure response body size, less then 42 bytes is ok.
How can I get bigger response? thanks.

tve, 2016-02-20:
Currently responses are limited to 100 chars in esp-link. I'm afraid the REST stuff needs work if the goal is to get sizable responses. It was written primarily to be able to post measurements to thingspeak or similar services.

rupin, 2016-06-28:
Facing a similar problem, responses to be received from openweather api are greater than 100 characters.

rupin, 2016-07-20:
Not the right way to solve this problem, but we faced a similar issue where the 100 chars limit hit us. We created a small heroku instance that would just give us the data we want.  Our use case relies on openweather maps api to give us weather data about a city/location.

Code can be accessed here.
https://github.com/CuriosityGym/WeatherDataApi