jeelabs/el-client #19: EL-Client sync failed!

safari2, 2016-09-02:
Hi.
I've tried both El-client examples, all off them don't work (EL-Client sync failed!). 
My configuration:
1) PC windows 7
2) mosquitto 32bit 
3) Arduino UNO R3
4) ESP-01 connected to UNO via RX-TX
5) Virtual com-port created successfully, I can flash UNO from Arduino IDE, work good.
Arduino log:
EL-Client starting!
А .. >wEL-Client sync failed!
А .. >wEL-Client sync failed!
А .. >wEL-Client sync failed!
А .. >wEL-Client sync failed!

Mosquitto console log:
d:\Program Files (x86)\mosquitto>mosquitto.exe -v
1472834513: mosquitto version 1.4.10 (build date 24/08/2016 21:03:24.73) starting
1472834513: Using default config.
1472834513: Opening ipv6 listen socket on port 1883.
1472834513: Opening ipv4 listen socket on port 1883.
1472834520: New connection from 192.168.1.105 on port 1883.
1472834520: New client connected from 192.168.1.105 as esp-link (c0, k60).
1472834520: Sending CONNACK to esp-link (0, 0)
1472834581: Received PINGREQ from esp-link
1472834581: Sending PINGRESP to esp-link
1472834641: Received PINGREQ from esp-link
1472834641: Sending PINGRESP to esp-link

FernandoGarcia, 2016-12-07:
Check the baud rate and Fmt on microcontroller console in ESP link web interface.

It's should be equal on setup of your code, by default it's 115200 on example.

Fmt is 8N1.

FernandoGarcia, 2016-12-07:
Too don't forget to enable SLIP on serial port.

safari2, 2016-12-08:
Esp-link works good as tcp-wifi brige at the same time. It is not the problem with connection and baud rate. How to enable slip and where? On Arduino or esp-link?

CharlyBrok, 2016-12-08:
You can enable slip on esp-link in the top of REST/MQTT page.

tve, 2016-12-12:
The client sync has been completely redone. It does require some changes in your sketch, see the examples. Please reopen if it's not fixed.

closed 2016-12-12 by tve

lfrantzeskakis, 2017-10-31:
Hello, 
I have a strange issue with esp07 (512+512) and esp.sync()
I have tested all the possible baud rate and checked connections 
Every time i get EL-Client sync failed! but i see the message from the web interface μC so the μC is sending messages to the esp.
Do you have any suggestions.
μC output messages
EL-Client starting!
0 // this is a Serial.println(ok);
EL-Client sync failed!
0
EL-Client sync failed!

I changed the code but nothing changed because the problem is in esp.sync() .

tve, 2017-10-31:
Do you have SLIP enabled in esp-link? it's on one of the web pages. Sorry, at work and don't have a module to test with...

lfrantzeskakis, 2017-10-31:
Yes SLIP is enabled . Also my MQTT server is connected.
Maybe this is a problem with the module it self (esp07) . 
I have 3 of them and they have the same issue . 
I will test it with esp01 . The only problem is that i need external antenna . 
Now i tested with SoftwareSerial and it still the same . 
Its strange because the serial port of esp is receiving the println messages ... Maybe tomorrow i will check it again

lfrantzeskakis, 2017-10-31:
Thank you in advance☺
Your work is superb!

tve, 2017-10-31:
Make sure you are using the latest versions of EL-Client and esp-link. I did fix-up the sync. Pasting verbatim messages from the esp-link debug log would be helpful too. SoftwareSerial is quite unlikely to work and if you can see the regular printf messages the baud rate is OK.

lfrantzeskakis, 2017-10-31:
Im using v3.0.14-g963ffbb but i had the same issue with v2 .
Maybe i need to replace the library elclient for the arduino ide.
Do you have the zip file of the latest version?

dontsovcmc, 2018-01-11:
I got "EL-Client sync failed!" with Arduino Pro mini 3.3v.
SLIP enabled
Serial baudrate in Arduino is the same as in /console.html
MQTT broker got status "esp-link" messages, but not from SLIP

esp-link: v3.0.14-g963ffbb

I have checked serial line: compile SoftwareSerial example to Arduino and send-receive data with Telnet and see them on Serial port of Arduino.

lfrantzeskakis, 2018-01-11:
From my late knowledge the problem is in connections . For me the problem was the level shiftier from 5v to 3.3. Now its perfect and working as a charm. So i suggest you to check your connections with the board first (its better to make the board again form the start if its easy) .

bklippstein, 2019-03-22:
Same problem using v3.0.14-g963ffbb with Arduino Due & ESP-01.
SLIP is enabled, MQTT server is connected, baud rate and Fmt are 115200 / 8N1.
>> EL-Client sync failed!

I also tested a Arduino Mega & ESP-01 (with resistors as level shifter), the sync worked.

Ray-Lin, 2024-07-12:
Try to 'Reset esp-link' in 'Debug log' page of esp-link may be help!