jeelabs/el-client #26: Arduino IDE : multiple definition

Dolly-esp8266, 2016-11-13:
this product is compatible with Arduino?
Win 10
esp8266 Wemos D1 mini (esp-link)
arduino mega 2560
Arduino IDE error message: 
Arduino: 1.6.10 (Windows 10), Плата:"Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\libraries\ELClient\ELClient.cpp.o (symbol from plugin): In function `ELClient::write(unsigned char)':

(.text+0x0): multiple definition of `ELClient::write(unsigned char)'

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\sketch\ELClient.cpp.o (symbol from plugin):(.text+0x0): first defined here

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\libraries\ELClient\ELClient.cpp.o (symbol from plugin): In function `ELClient::write(unsigned char)':

(.text+0x0): multiple definition of `ELClient::write(void*, unsigned int)'

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\sketch\ELClient.cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\libraries\ELClient\ELClient.cpp.o (symbol from plugin): In function `ELClient::write(unsigned char)':

(.text+0x0): multiple definition of `ELClient::Request()'

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\sketch\ELClient.cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\libraries\ELClient\ELClient.cpp.o (symbol from plugin): In function `ELClient::write(unsigned char)':

(.text+0x0): multiple definition of `ELClient::init()'

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\sketch\ELClient.cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\libraries\ELClient\ELClient.cpp.o (symbol from plugin): In function `ELClient::write(unsigned char)':

(.text+0x0): multiple definition of `ELClient::ELClient(Stream*)'

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\sketch\ELClient.cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\libraries\ELClient\ELClient.cpp.o (symbol from plugin): In function `ELClient::write(unsigned char)':

(.text+0x0): multiple definition of `ELClient::ELClient(Stream*)'

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\sketch\ELClient.cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\libraries\ELClient\ELClient.cpp.o (symbol from plugin): In function `ELClient::write(unsigned char)':

(.text+0x0): multiple definition of `ELClient::ELClient(Stream*, Stream*)'

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\sketch\ELClient.cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\libraries\ELClient\ELClient.cpp.o (symbol from plugin): In function `ELClient::write(unsigned char)':

(.text+0x0): multiple definition of `ELClient::ELClient(Stream*, Stream*)'

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\sketch\ELClient.cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\libraries\ELClient\ELClient.cpp.o (symbol from plugin): In function `ELClient::write(unsigned char)':

(.text+0x0): multiple definition of `ELClient::DBG(char const*)'

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\sketch\ELClient.cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\libraries\ELClient\ELClient.cpp.o (symbol from plugin): In function `ELClient::write(unsigned char)':

(.text+0x0): multiple definition of `ELClient::crc16Add(unsigned char, unsigned int)'

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\sketch\ELClient.cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\libraries\ELClient\ELClient.cpp.o (symbol from plugin): In function `ELClient::write(unsigned char)':
[... 122 lines cut ...]

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\sketch\ELClientMqtt.cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\libraries\ELClient\ELClientMqtt.cpp.o (symbol from plugin): In function `ELClientMqtt::ELClientMqtt(ELClient*)':

(.text+0x0): multiple definition of `ELClientMqtt::publish(__FlashStringHelper const*, __FlashStringHelper const*, unsigned int, unsigned char, unsigned char)'

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\sketch\ELClientMqtt.cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\libraries\ELClient\ELClientMqtt.cpp.o (symbol from plugin): In function `ELClientMqtt::ELClientMqtt(ELClient*)':

(.text+0x0): multiple definition of `ELClientMqtt::publish(char const*, __FlashStringHelper const*, unsigned int, unsigned char, unsigned char)'

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\sketch\ELClientMqtt.cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\libraries\ELClient\ELClientMqtt.cpp.o (symbol from plugin): In function `ELClientMqtt::ELClientMqtt(ELClient*)':

(.text+0x0): multiple definition of `ELClientMqtt::publish(__FlashStringHelper const*, unsigned char const*, unsigned int, unsigned char, unsigned char)'

C:\Users\Angel\AppData\Local\Temp\build6729ec4d194702014f42af9fcc02e370.tmp\sketch\ELClientMqtt.cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

Используем библиотеку ELClient в папке: C:\Users\Angel\Documents\Arduino\libraries\ELClient (legacy)
exit status 1

FernandoGarcia, 2016-12-09:
The library should be included two times in your sketch or there two copies of elclient in your IDE folder.

tve, 2016-12-12:
The errors such as `(.text+0x0): multiple definition of `ELClient::write(unsigned char)'` indicate that you are somehow loading the ELClient library twice into your executable, which is not going to work. Please reopen this issue if you need further assistance.

closed 2016-12-12 by tve