jeelabs/el-client #27: error while compiling with platformio 3.1.0 on arduino mega

ramarro123, 2016-11-17:
here it's the log of build for ELClientSocket, doesn't seems to compile on platformio 3.1.0

```
.piolibdeps/ELClient/ELClientSocket.cpp: In member function 'void ELClientSocket::socketCallback(void*)':
.piolibdeps/ELClient/ELClientSocket.cpp:70:25: error: invalid conversion from 'char**' to 'void**' [-fpermissive]

resp->popArgPtr(&_data);
^
In file included from .piolibdeps/ELClient/ELClient.h:11:0,
from .piolibdeps/ELClient/ELClientSocket.h:15,
from .piolibdeps/ELClient/ELClientSocket.cpp:10:
.piolibdeps/ELClient/ELClientResponse.h:43:13: error:   initializing argument 1 of 'int16_t ELClientResponse::popArgPtr(void**)' [-fpermissive]
int16_t popArgPtr(void **data);
^
Compiling .pioenvs/megaatmega2560/lib/DallasTemperature_ID54/DallasTemperature.o
.piolibdeps/ELClient/ELClientWebServer.cpp:24:47: error: 'static' may not be used when defining (as opposed to declaring) a static data member [-fpermissive]
static ELClientWebServer * ELClientWebServer::instance = 0;
```

freechw, 2016-11-19:
Arduino IDE 1.6.6 too!

ramarro123, 2016-11-22:
umm, i think that this lib it's dead :( RIP :D

steamboating, 2016-11-26:
Same with the Arduino IDE 1.6.5
This is such a promissing library - pleaase help us!
C:\Program Files (x86)\Arduino\libraries\ELClient\ELClientSocket.cpp: In member function 'void ELClientSocket::socketCallback(void*)':
C:\Program Files (x86)\Arduino\libraries\ELClient\ELClientSocket.cpp:70:25: error: invalid conversion from 'char**' to 'void**' [-fpermissive]
   resp->popArgPtr(&_data);

steamboating, 2016-11-26:
This is the full error text:

```

Build-Optionen wurden verändert, alles wird neu gebaut
Verwende die Bibliothek ELClient im Ordner: C:\Program Files (x86)\Arduino\libraries\ELClient (legacy)

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++ -c -g -Os -Wall -Wextra -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard -IC:\Program Files (x86)\Arduino\libraries\ELClient C:\Users\Rainer\AppData\Local\Temp\build842921587693719677.tmp\tcp_client.cpp -o C:\Users\Rainer\AppData\Local\Temp\build842921587693719677.tmp\tcp_client.cpp.o 
In file included from C:\Program Files (x86)\Arduino\libraries\ELClient/ELClient.h:9:0,
                 from tcp_client.ino:7:
C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial.h:51:40: note: #pragma message: Serialbuffersize 124
 #pragma message ("Serialbuffersize 124")
                                        ^
tcp_client.ino:12:34: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
tcp_client.ino:56:97: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
tcp_client.ino:56:97: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
tcp_client.ino:56:97: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
tcp_client.ino:56:97: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
tcp_client.ino:56:97: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
tcp_client.ino:56:97: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
tcp_client.ino:56:97: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
tcp_client.ino:56:97: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
tcp_client.ino:56:97: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
tcp_client.ino:56:97: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
tcp_client.ino:56:97: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
tcp_client.ino:56:97: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
tcp_client.ino:56:97: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
tcp_client.ino:56:97: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
tcp_client.ino:56:97: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
tcp_client.ino:56:97: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
tcp_client.ino: In function 'void tcpCb(uint8_t, uint8_t, uint16_t, char*)':
tcp_client.ino:80:15: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++ -c -g -Os -Wall -Wextra -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard -IC:\Program Files (x86)\Arduino\libraries\ELClient -IC:\Program Files (x86)\Arduino\libraries\ELClient\utility C:\Program Files (x86)\Arduino\libraries\ELClient\ELClient.cpp -o C:\Users\Rainer\AppData\Local\Temp\build842921587693719677.tmp\ELClient\ELClient.cpp.o 
In file included from C:\Program Files (x86)\Arduino\libraries\ELClient\ELClient.h:9:0,
                 from C:\Program Files (x86)\Arduino\libraries\ELClient\ELClient.cpp:9:
C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial.h:51:40: note: #pragma message: Serialbuffersize 124
 #pragma message ("Serialbuffersize 124")
                                        ^
C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++ -c -g -Os -Wall -Wextra -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard -IC:\Program Files (x86)\Arduino\libraries\ELClient -IC:\Program Files (x86)\Arduino\libraries\ELClient\utility C:\Program Files (x86)\Arduino\libraries\ELClient\ELClientCmd.cpp -o C:\Users\Rainer\AppData\Local\Temp\build842921587693719677.tmp\ELClient\ELClientCmd.cpp.o 
In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:224:0,
                 from C:\Program Files (x86)\Arduino\libraries\ELClient\ELClientCmd.h:10,
                 from C:\Program Files (x86)\Arduino\libraries\ELClient\ELClientCmd.cpp:9:
C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial.h:51:40: note: #pragma message: Serialbuffersize 124
 #pragma message ("Serialbuffersize 124")
                                        ^
[... cut ...]
In file included from C:\Program Files (x86)\Arduino\libraries\ELClient\ELClient.h:11:0,
                 from C:\Program Files (x86)\Arduino\libraries\ELClient\ELClientSocket.h:15,
                 from C:\Program Files (x86)\Arduino\libraries\ELClient\ELClientSocket.cpp:10:
C:\Program Files (x86)\Arduino\libraries\ELClient\ELClientResponse.h:43:13: error:   initializing argument 1 of 'int16_t ELClientResponse::popArgPtr(void**)' [-fpermissive]
     int16_t popArgPtr(void **data);
             ^
Fehler beim Kompilieren.
```

steamboating, 2016-11-26:
I went back to this branch and it works for me on my Frankenstein DUE
https://github.com/jeelabs/el-client/tree/syncreset
I still hope that some "hero" can explain how I can get the new branch work or fix the possible problem in the newer branch.

steamboating, 2016-11-27:
Same with the Arduino IDE 1.6.13
I noticed that you are using a different Arduino Software example in your video. I looked through all your el client versions starting from el-client-2.2 beta - but found an led example only in the 3.0 alpha. In this the setup routine is completely different:
void setup()
{
  Serial.begin(115200);
  
  URLHandler *ledHandler = webServer.createURLHandler(F("/SimpleLED.html.json"));
  ledHandler->loadCb.attach(&ledPageLoadAndRefreshCb);
  ledHandler->refreshCb.attach(&ledPageLoadAndRefreshCb);
  ledHandler->buttonCb.attach(&ledButtonPressCb);

  esp.resetCb = resetCb;
  resetCb();
}

Maybe you can upload the old and functional example for us to play with?

steamboating, 2016-11-29:
I found out that my Windows Arduino IDE is a little picky with the void/char definition. I changed the librarys in this way and was able to run the MQTT example. Please notice that I have done this without any knowledge of the internal functionality - but it works for me.

ELClientResponse.h 
43:13 int16_t popArgPtr(void **data); changed to 43:13 int16_t popArgPtr(char **data); ELClientResponse.cpp 47 changed to 47 int16_t ELClientResponse::popArgPtr(char **data) 55 *data = _arg_ptr; changed to 55 *data = (char *)_arg_ptr; ELClientWebServer.cpp 24 static ELClientWebServer * ELClientWebServer::instance = 0; changed to 24 ELClientWebServer * ELClientWebServer::instance = 0; 51 static void ELClientWebServer::webServerPacketHandler(void * response) changed to 51 void ELClientWebServer::webServerPacketHandler(void * response) 270 uint32_t cb = &webServerCb; changed to 270 void *cb = &webServerCb; 361 hnd->loadCb( hnd->URL.c_str() ); 362 else 363 hnd->refreshCb( hnd->URL.c_str() ); changed to 361 hnd->loadCb((char *) hnd->URL.c_str() ); 362 else 363 hnd->refreshCb((char *) hnd->URL.c_str() ); ELClientRest.h 93 void *_data; changed to 93 char *_data;

steamboating, 2016-11-29:
The blink LED example will run after the following change - but with this you kill the web page feedback...

ELClientWebServer.h
30 void ledPageLoadAndRefreshCb(const char * url) changed to
30 void ledPageLoadAndRefreshCb(char * url)

38 void ledButtonPressCb(const char * btnId) changed to 
38 void ledButtonPressCb(char * btnId)

Additional I had to add in the Arduino 
setup()
pinMode(LED_PIN, OUTPUT);

steamboating, 2016-11-30:
@freechw and @ramarro123 
Have you had a chance to test my fixing ideas - it works for me. Would be great to get some feedback from other Arduino IDE users like me.

ramarro123, 2016-12-03:
@steamboating i am not using this library at all, esp-link it's not actively maintained (as you can see not even this project it's updated or receive attention from @tve that it's the owner)

he said that he is open to PR but he won't work anymore on esp-link, probably the same apply to this lib as well.

i am using esp arduino port, you have to write more code probably, but it's actively developed
https://github.com/esp8266/Arduino/

tve, 2016-12-03:
@ramarro123 I strongly disagree with your statement that esp-link is not actively maintained. Just because I'm not working on the pieces you care about doesn't mean it's abandon-ware! What have you contributed to the project? Why don't you fix the issues? How much are you paying for it that you have expectations?

ramarro123, 2016-12-03:
@tve i have not even try to flash my esp with that, so well... i have pay 0, and i will continue with that 0 forever :)

also, about the state of the project, there are a lot of issue without reply (like this one, opened 16 days ago got a reply from you today after a ping)

esp-link codebase last update (and i am speaking about commit that doesn't touch docs or sub minor feature like adding baudrate on a select box) are quite old.

That's what i mean with "not actively maintained" ... something that it's not... actively maintained :)
i haven't say "abandon-ware", that's for dos games, your work it's still not old enough :)

And, about contribution, well, i can't /wont contrib to all project that i use (not to mention the one that i don't use because don't fit my project... and for esp-link, you miss stk500v2 and the lib doesn't compile in patformio out of the box to make me want to invest time on that).

Are you a committer of linux kernel (in case you use linux) or to any of the 1230210301230 GNU prog that you use? (including less or tail) 

Don't expect that people want to contrib to your work, nor conclude with a "do a pr" when someone signal an issue, it's your project not mine, if i will eventually do something of nice, i won't expect that other people help me (but i will be glad to, just... don't expect it)

Having said so, i repeat, your project it's not actively maintained, this lib it's old of 1 month and doesn't compile out of the box. That's facts, i was just replying to the other user that i haven't time/will to test his fixes (that are btw correct, the problem it's just the new compiler, you can use platform io probably adding the permissive switch on compiler)

steamboating, 2016-12-04:
@tve thank you Thorsten for sharing your work with the github world...
Rainer

tve, 2016-12-04:
@steamboating thanks & you're welcome. I'll try to get to this this week-end, but I'm already a bit over-committed and some of the code is not mine, so it'll take a bit more to double-check everything. Some of your changes don't look 100% correct but they should all work.

tve, 2016-12-12:
Mhh, weird, I just tried to reproduce this issue. I downloaded the latest arduino software, which is now 1.6.13, and I ran the Makefile for the MQTT example and while it spat out some warnings it worked great. This is all on Linux. The compiler flags are:
```
/home/arduino/arduino-1.6.13/hardware/tools/avr/bin/avr-g++   -Os -Wall -fno-exceptions -ffunction-sections -fdata-sections -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -fpermissive -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=150 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -DUSB_VID= -DUSB_PID= -I. -Iutil -Iutility -I /home/arduino/arduino-1.6.13/hardware/arduino/avr/cores/arduino -I /home/arduino/arduino-1.6.13/hardware/arduino/avr/variants/standard/ -I ../../../ELClient -I ../../../ELClient/utility -I ../../../ELClient/src  -c -o .lib/../../../ELClient/ELClient.cpp.o ../../../ELClient/ELClient.cpp
```
Sounds like I'm gonna have to install the IDE on Windows to try to reproduce this.

steamboating, 2016-12-13:
Hello Thorsten, thank you for looking over the code again - I have tested it on two different Windows PC's - win7 and 8 with the different available Arduino IDEs mentioned above with the out of the box settings. I thought already that you must have used an other setup for testing... Hopefully this is only a Windows problem and not an additional problem because of the installed boards? I am using the out of the box Arduino DUE settings.

tve, 2016-12-14:
Looks like I need some help with this :-). I went to https://www.arduino.cc/en/Main/Software and downloaded the current IDE for Windows. I then downloaded the current release of ELClient from https://github.com/jeelabs/el-client/releases/tag/v3.0.alpha0 (I picked the source zip archive). I unpacked the zip and copied the ELClient directory into the arduino libraries directory (C:\Program Files (x86)\Arduino\libraries). I then started the IDE, used the File>Open menu and openened the mqtt example sketch (C:\Program Files (x86)\Arduino\libraries\ELClient\examples\mqtt\mqtt.ino), selected the Arduino Mega ADK board and clicked on the checkmark button ("verify"). That happily compiled and linked the sketch. I tried the same with the webserver_led sketch with the same result.
There are some missing type-casts in the ElWebserver class, which I'm not so happy about, and I tried a pass at changing them but that resulted in a mess. I'll try again, but I don't see why this is blocking you guys. If someone has steps to reproduce the problem for me I'll gladly look into them.

tve, 2016-12-14:
I did another pass at fixing the various type cast warnings. This is pushed to master.

steamboating, 2016-12-14:
Thank you Thorsten, i like this kind of support - sorry? that you could not find "our" bug - I will try to find also why I have the problem and you don't - will also happely test the new version soon and tell my experience here.

Thanks for all your time!

steamboating, 2016-12-18:
Hello Thorsten - you did a great job - all errors messages are gone on my system with the update. 

I am using it on a Windows 8 System with Atmel Studio 7 and a self made Arduino DUE SAM3X8 PCB on which I have hard wired the Atmel with the ESP over Serial3 while doing all the debuging via SerialUSB.

In the moment I am working with the TCP demo - now i am able to make a server connection. But it took ME some time to find out that I have to switch "off" the "DEBUG LOG"  in the web interface because this disturbed/overloaded the data flow for the Serial3 communication. I am sure that you have written this in some of your great work - it would be nice if you could include this hint direct in the Arduino example code with the next update.

Switching off the DEBUG LOG also makes the LED example more stable and reliable because of the same reason.

Thanks for all your time you are inversting in this project!

steamboating, 2016-12-18:
An other hint for your el-web-led example:
The Atmel Studio and the Windows IDE "do not like" your direct read of the Led pin - this example always gives back the same state to the web interface. 
I changed the example in this way - maybe also worth a change in you next example update - no hurry with that...

bool LedStatus = false;
//==============================================
void ledPageLoadAndRefreshCb(char * url)
{
   if(LedStatus)
   webServer.setArgString(F("text"), F("LED ist an"));
   else
   webServer.setArgString(F("text"), F("LED ist aus"));
}
//=================================================
void ledButtonPressCb(char * btnId)
{
   String id = btnId;
   if( id == F("btn_on") ){
      digitalWrite(LED_PIN, true);
      LedStatus = true;
   }
   else if( id == F("btn_off") ) {
      digitalWrite(LED_PIN, false);
      LedStatus = false;
   }
}

tve, 2016-12-19:
Thanks for the update, I'll have to look into the LED issue.