jeelabs/el-client #28: Change message limit

FernandoGarcia, 2016-12-07:
Hi @tve !

Firstly thanks for this library.

Where can I edit the limit for message size?

I can see this message on log:

```
MQTT Client: Received topic: ******/********/topic/command, data: 20,1,03,00,10,00,1,2,3,4,5,6,7,33,2,24,1,1,K
MQTT: Data cb=0x00000d00 topic=*******/********/topic/command20,1,03,00,10,00,1,2,3,4,5,6,7,33,2,24,1,1,K len=44
```

**** are private infos.

With messages smaller it's working very well.

tve, 2016-12-08:
https://github.com/jeelabs/el-client/blob/master/ELClient/ELClient.h#L128
I hope this helps!

FernandoGarcia, 2016-12-09:
Hi @tve!

Thanks for your answer.

I changed it before to 350 and now to 550 but doesn't solve.

There another suggestion?

Best regards.

FernandoGarcia, 2016-12-09:
It's working with buffer size of 128 but only with baud rate at 9600 I don't know the reason, maybe because I'm using jumper wires of 30 cm.

For smaller messages works with baud rate at 115200.

Thanks for your time.

closed 2016-12-09 by FernandoGarcia