jeelabs/el-client #29: resetCB is not being called

FernandoGarcia, 2016-12-09:
Hi @tve !

The resetCB callback is not being called with esp-link 3.0 alpha0.

I'm testing with reset via  web interface and  push button.

Only to make sure there no error in my code here is:

```
#include <ELClient.h>
#include <ELClientMqtt.h>
#include <ELClientCmd.h>

const char *Username  = "*****";
const char *APIKEY = "******";

ELClient ESP8266(&Serial2);
ELClientMqtt MQTT(&ESP8266);

unsigned long last_millis;

boolean connected = false;

void wifiCb(void* response)
{

  ELClientResponse *res = (ELClientResponse*)response;
  if (res->argc() == 1)
  {
    uint8_t status;
    res->popArg(&status, 1);

    if (status == STATION_GOT_IP)
    {
      Serial.println("WIFI connected!");
    }
    else
    {
      Serial.print("Cant' connect to WIFI network: ");
      Serial.println(status);
    }
  }
}

void mqttConnected(void* response)
{
  char SUB_TOPIC[50];
  char CLIENT_ID[20];

  Serial.println("MQTT connected!");

  strcpy(SUB_TOPIC, Username);
  strcat(SUB_TOPIC, "/");
  strcat(SUB_TOPIC, APIKEY);
  strcat(SUB_TOPIC, "/topic/command");

  strcpy(CLIENT_ID, "ID: ");
  strcat(CLIENT_ID, Username);

  Serial.print(F("Command: ")); // Responde aos comandos
  Serial.println(SUB_TOPIC);

  MQTT.subscribe(SUB_TOPIC);
  connected = true;
}

void mqttDisconnected(void* response)
{
  Serial.println("MQTT disconnected");
  connected = false;
}
void mqttData(void* response)
{
  ELClientResponse *res = (ELClientResponse *)response;

  //Serial.print("Topic: ");
  String topic = res->popString();
  //Serial.println(topic);

  Serial.print("Message: ");
  String message = res->popString();
  Serial.println(message);
}

void resetCb()
{
  boolean ok = false;
  byte i = 0;

  Serial.println("Syncing or resyncing Arduino and ESP8266...");

  while ((ok == false) && (i < 10))
  {
    ok = ESP8266.Sync();
    if (!ok)
    {
      Serial.println("Sync failed!");
      i++;
    }
  }
  if (ok == true)
  {

    Serial.println("Synced!");

    MQTT.connectedCb.attach(mqttConnected);
    MQTT.disconnectedCb.attach(mqttDisconnected);
    MQTT.dataCb.attach(mqttData);
    MQTT.setup();
  }
}

void setup()
{
  Serial.begin(38400);
  Serial2.begin(9600);

  while (!Serial)
  {
    // Nothing to do. :)
  }
  Serial.flush();
  Serial.println();
  Serial.println("Setup...");

  ESP8266.wifiCb.attach(wifiCb); 
  ESP8266.resetCb = resetCb;
  resetCb();
}

void loop() 
{
  ESP8266.Process();

  char PUB_TOPIC[50];

  if ((connected == true) && ((millis() - last_millis) > 15000))
  {
    strcpy(PUB_TOPIC, Username);
    strcat(PUB_TOPIC, "/");
    strcat(PUB_TOPIC, APIKEY);
    strcat(PUB_TOPIC, "/topic/response");

    Serial.print(F("Response: ")); 
    Serial.println(PUB_TOPIC);
    Serial.println("Publishing...");
    char buf[12];

    itoa(millis() / 1000, buf, 10);
    MQTT.publish(PUB_TOPIC, buf);
    Serial.println("Published!");
    last_millis = millis();
  }
}
```

After a reset the Arduino crashes trying publish a message.

Best regards.

tve, 2016-12-09:
mhhh, can you try the syncreset branch of el-client, it looks like I forgot to merge that into master, ooops!

FernandoGarcia, 2016-12-09:
This code is crashing in infinity loop at Sync() with this branch.

tve, 2016-12-09:
Ugh, any details on the crash? And did you do this:
> Important: For this sketch to work you must turn off the UART debug log in esp-link (on the Debug Log page). The reason is that otherwise esp-link prints too much to its uart and then misses incoming characters.

FernandoGarcia, 2016-12-09:
With UART debug log in auto the ESP crashes together with Arduino.

With UART debug log disabled I have only this on serial monitor in loop:

```
Setup...
Syncing or resyncing Arduino and ESP8266...
```

On  debug log all messages like this:

```
 98691> cmdResponse: cmd=3 val=1212 argc=1
 98721> SLIP: start or end len=0 inpkt=1
 98721> SLIP: start or end len=0 inpkt=1
 98721> SLIP: start or end len=10 inpkt=1
 98721> cmdParsePacket: cmd=1 argc=0 value=1212
 98721> cmdExec: Dispatching cmd=SYNC
 98722> cmdResponse: cmd=2 val=1212 argc=0
 98722> cmdAddCb: 'wifiCb'->0x4bc added at 0
 98722> cmdWifiCb: wifiStatus=2
 98722> cmdGetCbByName: cb wifiCb found at index 0
 98722> cmdResponse: cmd=3 val=1212 argc=1
 98722> SLIP: start or end len=0 inpkt=1
 98738> SLIP: start or end len=0 inpkt=1
 98739> SLIP: start or end len=10 inpkt=1
 98739> cmdParsePacket: cmd=1 argc=0 value=1212
 98739> cmdExec: Dispatching cmd=SYNC
 98739> cmdResponse: cmd=2 val=1212 argc=0
 98739> cmdAddCb: 'wifiCb'->0x4bc added at 0
 98739> cmdWifiCb: wifiStatus=2
 98740> cmdGetCbByName: cb wifiCb found at index 0
 98740> cmdResponse: cmd=3 val=1212 argc=1
 98757> UART framing error (bad baud rate?)
101368> HTTP GET /console/text: 200, 61ms, h=12792
101373> HTTP GET /console/text: 200, 67ms, h=14832
101974> HTTP GET /console/text: 200, 666ms, h=15232
104743> HTTP GET /console/text: 200, 3436ms, h=15608
105773> HTTP GET /console/text: 200, 64ms, h=15232
108025> HTTP GET /console/text: 200, 63ms, h=15232
109762> HTTP GET /console/text: 200, 4052ms, h=15608
111712> HTTP GET /console/text: 200, 63ms, h=15216
113399> HTTP GET /log/text: 200, 9ms, h=11128
113452> HTTP GET /log/dbg: 200, 62ms, h=14648
113453> HTTP GET /menu: 200, 63ms, h=14872
114790> HTTP GET /console/text: 200, 3138ms, h=15608
118056> HTTP GET /console/text: 200, 59ms, h=15624
118182> HTTP GET /log/text: 200, 67ms, h=15624
121337> HTTP GET /console/text: 200, 63ms, h=15624
124120> MQTT: Send type=PINGREQ id=0000 len=2
124651> MQTT: Recv type=PINGRESP id=0000 len=2; Pend type=NULL id=00
124712> HTTP GET /console/text: 200, 58ms, h=15224
128087> HTTP GET /console/text: 200, 3433ms, h=15616
131478> HTTP GET /console/text: 200, 66ms, h=15224
136838> HTTP GET /console/text: 200, 5425ms, h=15616
136947> HTTP GET /console/text: 200, 54ms, h=15624
140181> HTTP GET /console/text: 200, 65ms, h=15624
143759> HTTP GET /console/text: 200, 59ms, h=15216
147337> HTTP GET /console/text: 200, 3635ms, h=13184
147352> HTTP GET /log/text: 200, 66ms, h=15224
150728> HTTP GET /console/text: 200, 65ms, h=14832
153631> HTTP GET /log/dbg: 200, 6345ms, h=15216
153775> HTTP GET /console/text: 200, 3111ms, h=15608
157790> HTTP GET /console/text: 200, 62ms, h=15232
```

[image]

tve, 2016-12-12:
As far as I can tell things are failing because you're trying to run the link with esp-link at 9600 baud. Esp-link busy-waits when transmitting to the attached uC and for long messages 9600 baud will block it for too long. The crash you're noticing most likely is a watchdog timer reset.
I would like to add interrupt driven TX to esp-link, but that's not the way it currently is.

FernandoGarcia, 2016-12-12:
Hi!

The problem is not the baud rate.

I found a strange behavior with your test code.

When the Serial for debug is given the code works with baud rate at 115200 or 9600.

`ELClient esp(&Serial2, &Serial);`

Remove the Serial for debug and it won't sync anymore and sometimes will crash.

`ELClient esp(&Serial2);`

Best regards.

FernandoGarcia, 2022-04-21:
Hi!

Revising this subject what I can say is:

With esp-link `v3.0.14` changing `ESP_TIMEOUT` to 4s I can receive messages of 454 byte with baud rate at `115200`.

But there's sign of reset callback on serial using `ELClient esp(&Serial3, &Serial);`

Meantime I can see this message with baud rate at `38400` and timeout at 5s:

```
ELC: got 20 @907: B04C 53868684 6087 F2 87 CF DF AA 28 87 85 C2 95 E3 40
ELC: Invalid CRC
ELC: got 15 @907: 808 A1A92929 69B9 E4 A4 84 C6 A4 84 85
ELC: Invalid CRC
ELC: got 600 @907: 6385 FC43FFE4 A0EC 24 E5 42 28 D5 21 29 A9 A1 C8 48 AF 25 F9 8E 88 48 F8 8F 66 E9 A8 EC 85 86 C4 86 62 E6 46 28 D1 27 6A A8 A0 72 6E AA 20 84 44 F2 84 66 E8 A3 E6 F1 84 86 46 A3 FF 84 86 46 42 A2 87 A4 85 85 A6 72 43 60 A6 24 84 E6 40 E6 21 21 21 21 29 2D 37 8E 80 20 46 40 8D 88 88 88 E8 4 62 62 84 1 82 C1 5E 24 76 23 22 60 87 A9 E4 C5 28 32 BC 52 E4 24 86 84 E7 A4 85 DF AF 30 28 28 24 C2 A6 A4 62 FA F7 E6 84 C6 42 65 43 FD 2E 28 24 A6 43 FD 2E 28 24 A6 43 3C 2E 68 24 CC 61 A4 29 20 26 26 E4 30 EA 72 43 DE C4 87 85 80 40 23 28 2A A8 60 2C 28 2A EE 82 C6 29 A A A D A 20 20 20 31 32 32 3E 20 D A 20 20 20 31 32 33 3E 20 2A 2A 20 65 73 70 2D 6C 69 6E 6B 20 76 33 2E 30 2E 31 34 2D 67 39 36 33 66 66 62 62 D A 20 20 20 31 32 33 3E 20 46 6C 61 73 68 20 63 6F 6E 66 69 67 20 72 65 73 74 6F 72 65 20 6F 6B D A 20 20 20 31 32 33 3E 20 43 4F 4E 4E 20 6C 65 64 3D 30 D A 20 20 20 31 32 33 3E 20 53 45 52 20 6C 65 64 3D 31 34 D A 20 20 20 32 32 32 3E 20 57 69 66 69 20 69 6E 69 74 2C 20 6D 6F 64 65 3D 53 54 41 D A 20 20 20 32 32 32 3E 20 57 69 66 69 20 75 73 65 73 20 44 48 43 50 2C 20 68 6F 73 74 6E 61 6D 65 3D 65 73 70 2D 6C 69 6E 6B D A 20 20 20 32 32 32 3E 20 22 69 70 22 3A 20 22 30 2E 30 2E 30 2E 30 22 D A 20 20 20 32 32 32 3E 20 22 6E 65 74 6D 61 73 6B 22 3A 20 22 30 2E 30 2E 30 2E 30 22 D A 20 20 20 32 32 38 3E 20 22 67 61 74 65 77 61 79 22 3A 20 22 30 2E 30 2E 30 2E 30 22 D A 20 20 20 32 33 36 3E 20 22 68 6F 73 74 6E 61 6D 65 22 3A 20 22 65 73 70 2D 6C 69 6E 6B 22 D A 20 20 20 32 34 35 3E 20 73 6C 65 65 70 20 65 6E 61 62 6C 65 2C 74 79 70 65 3A 20 32 D A 20 20 20 32 35 32 3E 20 48 74 74 70 64 20 69 6E 69 74 2C 20 63 6F 6E 6E 3D 30 78 33 66 66 66 33 38 32 30 D A 20 20 20 32 36 32 3E 20 4E 6F 20 75 73 65 72 20 66 69 6C 65 20 73 79 73 74 65 6D 20
ELC: Invalid CRC
```

Looks like the reset callback message is very large or there's another problem.

Best regards.