jeelabs/el-client #30: How to get MQTT status

FernandoGarcia, 2016-12-14:
Hi @tve !

Sorry by bother you again.

I would like to know if there any function to get MQTT status before try publish a message.

I'm using mqtt.connectedCb callback but it is not enough.

Best regards.

tve, 2016-12-15:
There is an mqtt Disconnected Callback that will tell you when the TCP connection dies. The Wifi status callback will tell you about the wifi association. There really isn't that much more info available...

FernandoGarcia, 2016-12-15:
Hi!

Disconnected callback is not called after a esp-link reset too. 

It works rarely.

Best regards.

tve, 2016-12-16:
After an esp-link reset you get a reset callback... See https://github.com/jeelabs/el-client/blob/master/ELClient/ELClient.h#L119-L121
There are different cases and they require slightly different handling. It's not as simple as I wished it to be, but I don't see a way to make it simpler...

FernandoGarcia, 2016-12-16:
Hi!

Thanks for your answer!

My solution was call [this functions](https://github.com/FernandoGarcia/Ferduino_with_webcontrol_beta/blob/master/Ferduino_with_webcontrol_beta/ESP8266.ino#L52) always before try publish something.

All callbacks are not working fine to me after a reset.

Best regards.

FernandoGarcia, 2016-12-23:
Hi!

A function to be used like this could be very interesting.

`if (!MQTT.connected())`

Best regards.