jeelabs/el-client #36: Invalid CRC

joseamirandavelez, 2017-07-15:
I am using EL-Client in an Arduino 101. I get the following error:

```
EL-Client starting!
Serial communication enabled...

ELC: got 128 @A800EE1C: 3120 203E3436 3331 53 4C 49 50 3A 20 73 74 61 72 74 20 6F 72 20 65 6E 64 20 6C 65 6E 3D 30 20 69 6E 70 6B 74 3D 30 D A 20 31 31 33 36 34 3E 20 53 4C 49 50 3A 20 73 74 61 72 74 20 6F 72 20 65 6E 64 20 6C 65 6E 3D 30 20 69 6E 70 6B 74 3D 31 D A 20 31 31 33 36 35 3E 20 53 4C 49 50 3A 20 73 74 61 72 74 20 6F 72 20 65 6E 64 20 6C 65 6E 3D 31 30 20 69 6E 70 6B 74 3D 31 D A 20

ELC: Invalid CRC

ELC: got 51 @A800EE1C: 3120 203E3039 3431 63 6D 64 52 65 73 70 6F 6E 73 65 3A 20 63 6D 64 3D 32 20 76 61 6C 3D 2D 31 34 37 36 33 33 34 30 38 38 20 61 72 67 63 3D 30 D A
```

I have tried everything I know to get rid of it. The latest "fix" was to disconnect the serial communication for 5 seconds using an octocoupler while the ESP-12 boots to prevent the boot signal from interfering with the Arduino. It doesnt work. As soon as I try to attach to the wifi, it shows the error above. I am using the [REST](https://github.com/jeelabs/el-client/tree/master/ELClient/examples/rest) example.

saspol, 2017-09-03:
Have same problem.
Tried on esp-link v2.2.3 and v3.0.14, always got CRC error with empty 'topic' and 'data'.
Monitoring with desktop MQTT client shows everything fine.

```
EL-Client synced!
ConnectedCB is 0x48E
EL-MQTT ready
ELC: got 128 @55B: 3120 203E3139 3234 31 34 34 32 31 3E 20 53 4C 49 50 3A 20 73 74 61 72 74 20 6F 72 20 65 6E 64 20 6C 65 6E 3D 33 31 20 69 6E 70 6B 74 3D 30 D A 20 31 34 34 32 31 3E 20 53 4C 49 50 3A 20 73 74 61 72 74 20 6F 72 20 65 6E 64 20 6C 65 6E 3D 33 34 20 69 6E 70 6B 74 3D 31 D A 20 31 34 34 32 31 3E 20 63 6D 64 50 61 72 73 65 50 61 63 6B 65 74 3A 20 63 6D 64 3D 31 30 20 61 72 67 63
ELC: Invalid CRC
ELC: got 10 @55B: 3 48E 0 9D 6A
RESP_CB: 1166 0
MQTT connected!
ELC: got 128 @55B: 3120 203E3435 3534 53 4C 49 50 3A 20 73 74 61 72 74 20 6F 72 20 65 6E 64 20 6C 65 6E 3D 37 35 20 69 6E 70 6B 74 3D 30 D A 20 31 34 35 35 39 3E 20 53 4C 49 50 3A 20 73 74 61 72 74 20 6F 72 20 65 6E 64 20 6C 65 6E 3D 33 30 20 69 6E 70 6B 74 3D 31 D A 20 31 34 35 35 39 3E 20 63 6D 64 50 61 72 73 65 50 61 63 6B 65 74 3A 20 63 6D 64 3D 31 32 20 61 72 67 63 3D 32 20 76 61 6C 75
ELC: Invalid CRC
ELC: got 42 @55B: 3 4A0 2 18 0 2F 68 6F 6D 65 2F 62 6C 75 65 2F 73 70 6F 74 6C 69 67 68 74 41 2F 6F 6E 0 0 1 0 31 0 DE 5D
RESP_CB: 1184 2
Received: topic=
data=
```

saspol, 2017-09-04:
I found problem is in ```mqttData``` callback function.
In my case i have inside many checking like this:
```
if ( topic == "/home/blue/spotlightA/on") {
    spotlightA.setBrightness(0);
  } else
  if ( topic == "/home/blue/spotlightA/off" ) {
    spotlightA.setBrightness(255);
  } else
  if ( topic == "/home/blue/spotlightB/on" ) {
    spotlightB.setBrightness(0);
  } else
<...>
```
So, if i have 12 ```if..then..else``` everything working as expected. Just one more add in any place -  ```topic``` and ```data``` always become empty. Again coment out any of them - everything working fine.
What is this? How to overcome this behaviour??

sejgit, 2021-01-28:
I know this has been quiet for a bit ; anyone figure out the work-around.
Get below when I send was seems to be the right JSON through MQTT.
This works on another microcontroller ESP32/8266 when not using el-client.

ELC: got 128 @5A5: 3 4EA 2 16 0 73 65 6A 2F 73 68 65 6C 66 73 74 72 69 70 2F 63 6F 6E 74 72 6F 6C 72 0 7B A 20 20 22 73 74 61 74 65 22 3A 31 30 30 2C A 20 20 22 62 72 69 67 68 74 6E 65 73 73 22 3A 31 30 30 2C A 20 20 22 63 6F 6C 6F 72 22 3A 7B A 20 20 20 20 22 72 22 3A 30 2C A 20 20 20 20 22 67 22 3A 30 2C A 20 20 20 20 22 62 22 3A 30 2C A 20 20 20 20 22 77 22 3A 31 30 30 A
ELC: Invalid CRC

markshap, 2022-08-03:
> I know this has been quiet for a bit ; anyone figure out the work-around. 

It looks like the debug logging interferes with the data exchange. Turning off the UART debug log on the esp-link web console "debug log" page fixes it.

[image]

Kzibi, 2023-05-19:
> > I know this has been quiet for a bit ; anyone figure out the work-around.
> 
> It looks like the debug logging interferes with the data exchange. Turning off the UART debug log on the esp-link web console "debug log" page fixes it.
> 
> [image]

It does not

bhunting, 2024-02-09:
Try lowering your baud rate to 19200, or slower, in the microcontroller program and on the uc Console tab of the esp-link web page.
I have multiple esp-link systems running and my systems with the esp-01 "remote", maybe 5 inches of flying leads away from the uC, I get Invalid CRC errors in the uC Console window and the uC won't attach to the esp-link properly.  If I set the serial port baud rate to 19200 it works.