jeelabs/el-client #39: binary data over UDP is truncated at first \0 byte

zerorist, 2017-11-11:
Hello
I found that using 
`void ELClientSocket::send(const char* data, int len)`
with any string containing \0 leads to the sending of a truncated string.
I don't understand well the lines 
`       _elc->Request(data, strlen(data));
	if (data != NULL && len > 0)
	{
		_elc->Request(data, len);
	}`
but I tried to comment out the first line and it seems to be functionnal.
Can you please confirm I didn't make a mistake, and eventually patch this class?
Your libs are really usefull to me, thank you a lot!
Thanks you for your attention.

Yaonian72, 2021-06-14:
After sending about 24 hours data, the wifi loop will stop and sending nothing.

Yaonian72, 2021-06-14:
I also have a similar situation. After sending about 24 hours of data, the wifi loop will send nothing.