jeelabs/el-client #46: How to program on elclient?

Bagunda, 2018-10-15:
I don't understand.
How to program on elclient?
I write firmware, web interface is work.
If I write 'blink' sketch from Arduino IDE, then nothin works.
I have search over 300 sites. And nothing

rtek1000, 2018-11-17:
I'm just starting with the el-client software, but from what I understand:

1- You need to leave esp8266 running with esp-link, 
[image]
[https://github.com/jeelabs/esp-link](url)

2- You need to enable the SLIP function (on the tab below the services called "REST & MQTT"): "[x] Enable SLIP on serial port". Available on the esp-link settings page.
[image]

3- The speed (Baud) of the port of the "uC console" must be with the same speed of the program el-client.
[image]

4- So you can to program the Arduino with the el-client example code.
[image]
[https://github.com/jeelabs/el-client/tree/master/ELClient/examples](url)

5- The communication functions are similar to the Arduino programs for esp8266, the difference is that an intermediate is being used between the Arduino and esp8266, which is the esp-link firmware.

> ELClient:  is a Wifi library for arduino that uses the SLIP protocol to communicate via serial with an ESP8266 module running the esp-link firmware.
[image]

[https://github.com/jeelabs/el-client](url)

[image]

6- I believe it would be advisable to first learn to use esp8266 directly with the Arduino.
[image]

Getting Started with ESP8266 and Arduino: ESP8266 Arduino Interface:
[https://www.electronicshub.org/esp8266-arduino-interface/](url)

If the Atmega328 (like an Arduino Pro Mini) has the UNO bootloader, the IDE can work with the "Arduino WiFi" option, to program the Atmega328 using the esp-link as a bridge. You need to configure the mDNS service with the name "arduino", details in:
[https://github.com/jeelabs/esp-link/issues/394](url)

[image]

deadprogram, 2019-01-14:
@rtek1000 that is a great summary. Thank you.