jeelabs/el-client #57: rest.ino: "EL-Client sync failed!"

sixtyfive, 2020-11-29:
Hi! So I'm trying to get SLIP working and have been playing with the `rest.ino` example. Unfortunately, the only thing that comes out of it is:

```
��␁␀␀␀�␄␀␀�6�EL-Client sync failed!
```

My serial speed is set to `57600`, which is also the speed set at `/console.html` and the one I have to use for uploading (for some reason, 115200 isn't working. As that sketch explicitly mentions that the serial speed is critical to making SLIP work: is there another setting somewhere that need to be changed? If not, what else could I being doing wrong, given that both uploading sketches as well as the remote serial console are working fine and just SLIP is not?

acejoh, 2021-06-23:
Did you enable SLIP on the services page?

sixtyfive, 2021-06-23:
Yes, I wouldn't have written an Issue if I hadn't. By now I've forgotten what project this was for, though, and it'd probably be best to try and reproduce with the current master and then just close this Issue. It'll be a few weeks before I will make it back to the bench, so feel free to just close it now.